gaoxiang12/ORB-YGZ-SLAM #2: Compile failed in Nvidia jetson TX1.

hajlyx, 2017-07-06:
Does this project only support Intel x86-64 architecture?

gaoxiang12, 2017-07-06:
Sorry I haven't tested it in arm architecture. Since it has some SSE accelerated codes so I'm afraid that there will be some problems about them.

hajlyx, 2017-07-06:
I just delete CMake_CXX_FLAGS such as "msse","msse2" and so on in CMakeList.txt and it's worked.
I test it with TUM dataset named "rgbd_dataset_freiburg3_long_office_household".
The fps here is almost 12-15,which is faster then original ORB-SLAM2.
By the way,does this project have a pointcloud map viewer like your previous project ORBSLAM2_with_pointcloud_map?
I tested previous project on Nvidia TX1 but it's too slow and crashed after running five or less seconds.