gaoxiang12/ORB-YGZ-SLAM #4: Config file incorrect

doomzzju, 2017-07-08:
Hi guys,
Thank you for sharing your code! But maybe you put a wrong config file in the repo....(no IMU params). Looking for the update.

Regards

gaoxiang12, 2017-07-10:
Thank you. We write the IMU parameters directly in src/IMU/configparams.h and you need to rewrite them when using a different IMU.

doomzzju, 2017-07-10:
Opps, my fault, not IMU params, is extrinsic calib and bUseIMU? In fact it does work when using VIOrb's config, but the original config in this repo is incorrect. It not a big deal, but need to be fix, right?

mail4pawan2003loud, 2017-09-03:
Dear Gaoxiang,
Could you please share your YAML configparams file so that I can run and test the EuRoC dataset in VIO mode ? thanks in advance.

heiitu, 2018-01-15:
i found the same problems XD