gaoxiang12/ORB-YGZ-SLAM #5: can not work with the TUM dataests

wowdeath, 2017-07-10:
When i run with the TUM dataests, it can start but it will immediately exit and the terminal show " Illegal instruction (core dumped)" .I want to know why did this happen and what can i do to solve it?
Thank you very much.

gaoxiang12, 2017-07-10:
Do you give the correct paths? Are the images correctly loaded?

wowdeath, 2017-07-11:
Yes, the path is correct and the images can be used in ORB-SLAM2, but when i use it in ORB-YGZ-SLAM,it will  immediately exit and the terminal show " Illegal instruction (core dumped)" . Thank you very much for your answer.

factory-fish, 2020-04-23:
> When i run with the TUM dataests, it can start but it will immediately exit and the terminal show " Illegal instruction (core dumped)" .I want to know why did this happen and what can i do to solve it?
> Thank you very much.
Have you solved it ?

fanfanfan-fan, 2021-04-04:
> 当我使用TUM dataestest运行时，它可以启动，但是会立即退出，并且终端显示“非法指令（内核已转储）”。我想知道为什么会发生这种情况，我该怎么办？
> 非常感谢你。

我也是这样 ，请问您最后是跑的什么数据集能成功呢