gaoxiang12/ORB-YGZ-SLAM #7: Compile failed at examples (undefined reference to `cv::read...)

kuonangzhe, 2017-07-27:
Nice work! I wanted to try on the code but got error when compiling. Installation of project by running ./generate.sh.

System: Ubuntu 17.04
OpenCV: 3 from github master 
dependency all installed

The error as follows:

[ 65%] Built target ygz-SLAM
[ 68%] Linking CXX executable ../Examples/Monocular/mono_tum
[ 70%] Linking CXX executable ../Examples/Monocular/mono_euroc
[ 75%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 75%] Linking CXX executable ../Examples/Monocular/mono_kitti
../lib/libygz-SLAM.so: undefined reference to `cv::read(cv::FileNode const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:153: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
../lib/libygz-SLAM.so: undefined reference to `cv::read(cv::FileNode const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
../lib/libygz-SLAM.so: undefined reference to `cv::read(cv::FileNode const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_euroc.dir/build.make:153: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
CMakeFiles/mono_tum.dir/build.make:153: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
../lib/libygz-SLAM.so: undefined reference to `cv::read(cv::FileNode const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:153: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Thank you so much!

kuonangzhe, 2017-07-27:
I also compiled the original ORB-SLAM2 and everything went well with the same system setting.

gaoxiang12, 2017-07-28:
looks like your binary files cannot link the OpenCV libraries. I updated the CMakeLists.txt now, please retry it.

kuonangzhe, 2017-07-29:
@gaoxiang12 
Thanks for the help! Yeah the new CMakeLists.txt solved the problem! Looks like it is a good idea to add those lines to solve similar problems as "undefined reference to xxx"~

closed 2017-07-29 by kuonangzhe