gaoxiang12/ORB-YGZ-SLAM #9: Fail to run EuRoC example

kuonangzhe, 2017-07-29:
Thanks for the awesome work! The code is well written with high readability! 

Here is a problem for me. Though I built the project successfully, when running the example of EuRoC, I failed. 

My config: 
Ubuntu: 17.04
OpenCV: 3 from github master (after installing ros seems like the computer starts to use ros-lunar-opencv3, as in error shows below)
ROS: Lunar (but I did not use ROS)

Here is the error:
czm@czm:~/SLAM/ORB-YGZ-SLAM$ ./Examples/Monocular/mono_euroc Vocabulary/ORBvoc.bin Examples/Monocular/EuRoC.yaml /home/czm/Dataset/EuRoC_MAV/mav0/cam0/data Examples/Monocular/EuRoC_TimeStamps/MH01.txt 
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0729 10:18:43.053556  3470 System.cc:107] This is YGZ-SLAM, forked from ORB-SLAM proposed by Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See ORB-SLAM's LICENSE.txt.

I0729 10:18:43.053719  3470 System.cc:114] Input sensor was set to: 
I0729 10:18:43.053740  3470 System.cc:117] Monocular
I0729 10:18:43.100296  3470 System.cc:132] 
Loading ORB Vocabulary. This should be super fast ...
I0729 10:18:43.571415  3470 System.cc:148] Vocabulary loaded!

Parameters: 
VINS initialize time: 0
Discart time in test data: 0
save tmp file in 
open rosbag: 
imu topic: 
image topic: 
local window size: 0
timestamp image delay to imu: 0
Tbc inited:
 1  0  0  0
 0 -1  0  0
 0  0 -1  0
 0  0  0  1
[1, 0, 0, 0;
 0, -1, 0, 0;
 0, 0, -1, 0;
 0, 0, 0, 1]
Tcb inited:
 1  0  0 -0
 0 -1  0  0
 0  0 -1  0
 0  0  0  1
[1, 0, 0, -0;
 0, -1, 0, 0;
 0, 0, -1, 0;
 0, 0, 0, 1]
Tbc*Tcb:
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
[1, 0, 0, 0;
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
whether acc*9.8? 0/1: 0
whether use IMU 0/1: 0

Camera Parameters: 
- fx: 458.654
- fy: 457.296
- cx: 367.215
- cy: 248.375
- k1: -0.283408
- k2: 0.0739591
- p1: 0.00019359
- p2: 1.76187e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 4
- Scale Factor: 2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
I0729 10:18:43.631961  3470 LocalMapping.cc:751] mnLocalWindowSize:0, mbUseIMU:0

-------
Start processing sequence ...
Images in the sequence: 3682

OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /tmp/binarydeb/ros-lunar-opencv3-3.2.0/modules/core/src/matrix.cpp, line 483
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-lunar-opencv3-3.2.0/modules/core/src/matrix.cpp:483: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

Aborted (core dumped)

Thanks for the help!

kuonangzhe, 2017-08-09:
Hi, I fixed the OpenCV issue from using ros-lunar-opencv to the normal opencv-master from github, by modifying the CMakeLists.txt: (I installed opencv in ~/opencv)

SET("OpenCV_DIR" "~/opencv")
find_package(OpenCV REQUIRED PATHS "~/opencv")
if(NOT OpenCV_FOUND)
   find_package(OpenCV 2.4.3 QUIET)
   if(NOT OpenCV_FOUND)
      message(FATAL_ERROR "OpenCV > 2.4.3 not found.")
   endif()
endif()
include_directories(${OpenCV_INCLUDE_DIRS})

While the result of running the EuRoC exmaple is still a failure: Segmentation fault (core dumped)
When running VINS demo of it, it shows again similar result:

imagestart:0
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /home/czm/opencv/modules/core/src/matrix.cpp, line 489
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/czm/opencv/modules/core/src/matrix.cpp:489: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

Aborted (core dumped)

I think there might be some bug with the code. I will also update the status when finding anything.

kuonangzhe, 2017-08-26:
@gaoxiang12 
After a bunch of debugging, I found that the main issue might come from the value initialization in frame due to some compiler differences. My friend's computer runs the code well with Ubuntu 16. But for me, I have to add one line
```
N = 0;
```
in src/frame.cc at line 282 after frame ID definition in monocular Frame part. The N = 0 will initialize the frame with value 0 in my circumstance, or it will become a randomly large number that destroy the program. In some system, this initialization is automatically done. 

Hope this could help with others if needed~

450932894, 2019-11-08:
Hi！
Did you find the ROS for ygz-slam? Can you share the url?
thank you！