gaoxiang12/ORB-YGZ-SLAM #11: -- Could NOT find CHOLMOD (missing:  CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES)

zhm1995, 2017-09-25:
`Build type: Release
-- Using flag -std=c++11.
-- Could NOT find CHOLMOD (missing:  CHOLMOD_INCLUDE_DIR CHOLMOD_LIBRARIES)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CHOLMOD_INCLUDE_DIR
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
   used as include directory in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM
CHOLMOD_LIBRARY
    linked by target "ygz-SLAM" in directory /home/sunkaikai/skk/zlB/ORB-YGZ-SLAM

-- Configuring incomplete, errors occurred!
`
would you please have a look? There might be some problems with g2o?

highlightz, 2017-10-18:
Did you install cholmod library? For example,  
`sudo apt-get install libcholmod2.1.2`

zhm1995, 2017-10-25:
really thank you for your help
I have figured out this problem

yaobaishen, 2018-02-07:
Hi,
I have seen the same error, could you share the solution? I install the libcholmod2.1.2 with no luck.

yaobaishen, 2018-02-07:
finally I sove it by: sudo apt-get install libsuitesparse-dev

zhm1995, 2018-02-07:
yeah... the same as me...

yaobaishen, 2018-02-07:
Sorry，but I see another issue:

slambook/ch10/g2o_custombundle/g2o_bundle.cpp:21:55: fatal error: g2o/solvers/cholmod/linear_solver_cholmod.h: No such file or directory
 #include "g2o/solvers/cholmod/linear_solver_cholmod.h"

zhm1995, 2018-02-07:
download the suitesparse package in http://faculty.cs.tamu.edu/davis/suitesparse.html

yaobaishen, 2018-02-07:
Thanks, after install the suitespase package I should rebuild the g2o library, then the ch10 example could run.

chunibyo-wly, 2020-12-18:
After I tried install dependencies in install_dependencies.sh, I couldn't find cholmod.h either.
However I can find this file in my `/usr/include/suitesparse/cholmod.h`, so I tried absolute path, It works.
Maybe it's a temporal solution.