gaoxiang12/ORB-YGZ-SLAM #13: when testing with IMU no certain files

zhm1995, 2017-10-25:
in localmapping.cc 
TryInitVIO:
fgw.open("/home/jp/opensourcecode/ORB_SLAM2/tmp/gw.txt");
fscale.open("/home/jp/opensourcecode/ORB_SLAM2/tmp/scale.txt");
fbiasa.open("/home/jp/opensourcecode/ORB_SLAM2/tmp/biasa.txt");
fbiasg.open("/home/jp/opensourcecode/ORB_SLAM2/tmp/biasg.txt");

how can I get gw.txt scale.txt biasa.txt?

highlightz, 2017-10-25:
Change the path to match your own machine, please. e.g. 
```
fgw.open("/home/YOUR_MACHINE/gw.txt");
```

zhm1995, 2017-10-25:
sorry.. probably I haven't spoken clearly..
I mean I find no where to download such files..

highlightz, 2017-10-25:
@zhm1995 These files are generated by this program when initialization is performed, so you do not need to download them.

zhm1995, 2017-10-25:
But I cannot find any code meaning generating such files..

TouqeerAhmad, 2017-10-26:
Essentially these files are not required, they are written once by the original author so that the gravity, scale and biases which are acquired during pre-integration of IMU can be used afterwards. I think you need not to worry about these.

zhm1995, 2017-10-29:
thank u so much..acutally I was encountered with this problem so I thought there was something wrong with such files..
Local Mapping RELEASE
VINS inited, quit VINS init thread
Quit VINSInitThread
I1029 13:13:38.487130  3878 Tracking.cc:2090] Failed. return back to feature methods
I1029 13:13:38.644781  3878 Tracking.cc:1591] Search by projection returns 94
I1029 13:13:38.689330  3878 Tracking.cc:502] Track Local map with imu failed.
I1029 13:13:38.689432  3878 Tracking.cc:531] Set vision weak = true
I1029 13:13:38.689631  3878 Tracking.cc:605] Try using imu predicted pose
W1029 13:13:38.785576  3878 Tracking.cc:2081] Last frame have less observations: 0, sparse alignment may have a erroneous result, return back to feature method.
I1029 13:13:38.914366  3878 Tracking.cc:1091] Search a wider window
I1029 13:13:38.914747  3878 Tracking.cc:1097] Match is not enough, failed.
I1029 13:13:38.919277  3878 Tracking.cc:531] Set vision weak = true
I1029 13:13:38.920372  3878 Tracking.cc:605] Try using imu predicted pose
W1029 13:13:39.013690  3878 Tracking.cc:2081] Last frame have less observations: 0, sparse alignment may have a erroneous result, return back to feature method.
I1029 13:13:39.148257  3878 Tracking.cc:1091] Search a wider window
I1029 13:13:39.149822  3878 Tracking.cc:1097] Match is not enough, failed.
I1029 13:13:39.171463  3878 Tracking.cc:1591] Search by projection returns 182
CHOLMOD warning: not positive definite
段错误 (核心已转储)

Did any one of you face with problem too? 

(testing with mono_euroc_vins)

TouqeerAhmad, 2017-10-29:
Yes, this is happening for me as well especially for my own data set once It goes for imu initialization thread and updates the scale and map afterwards these messages starts displaying and also the trajectory flies away.

May be someone else can comment on this.