gaoxiang12/ORB-YGZ-SLAM #19: 在编译时出现的关于sophus的问题，sophus已经安装并通过测试

StilesGao, 2018-01-22:
test_LK_tracking.cpp:(.text.startup+0x6f8)：对‘Sophus::SE3::SE3()’未定义的引用
test_LK_tracking.cpp:(.text.startup+0xee9)：对‘Sophus::SE3::SE3()’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SO3::log() const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::SE3(Sophus::SO3 const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::operator=(Sophus::SE3 const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::SE3(Sophus::SE3 const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::inverse() const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::log() const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::operator*(Sophus::SE3 const&) const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SO3::matrix() const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::matrix() const’未定义的引用
../../lib/libygz-algorithm.so：对‘Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const’未定义的引用
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_LK_tracking.dir/build.make:182: recipe for target '../bin/test_LK_tracking' failed
make[2]: *** [../bin/test_LK_tracking] Error 1
CMakeFiles/Makefile2:238: recipe for target 'test/CMakeFiles/test_LK_tracking.dir/all' failed
make[1]: *** [test/CMakeFiles/test_LK_tracking.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

gaoxiang12, 2018-01-23:
未链接sophus或sophus版本不正确。

StilesGao, 2018-01-25:
好的，谢谢您了，问题解决了

| |
邮箱：gxdbill123@163.com
|

签名由 网易邮箱大师 定制

在2018年01月23日 22:01，Xiang Gao 写道：

未链接sophus或sophus版本不正确。

kantlee, 2020-03-24:
请问你是如何解决的?

fanfanfan-fan, 2021-04-04:
请问您现在跑通这个ORB-YGZ-SLAM代码了吗 希望您能回复一下 非常感谢！